hku-mars/FAST-LIVO #125: 您好 打扰您了，我采用硬件设备为MID360+STM32做硬件同步，且使用串口助手检测PA9串口可以发出以下文本

wangdebin00223, 2024-09-07:
$GPRMC,000601.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2E
$GPRMC,000602.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2D
$GPRMC,000603.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2C
$GPRMC,000604.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2B
$GPRMC,000605.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2A
$GPRMC,000606.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A29
$GPRMC,000607.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A28
$GPRMC,000608.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A27
$GPRMC,000609.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A*26
但是我将STM32的PA9和PB5分别接入MID360线的10号和8号线，在LivoxViewer查看Time Sync Type :显示 No Sync
并且有警告。能麻烦您帮忙分析一下，是哪里的问题吗？卡在这里好久了，一直排查不出来,十分感谢!!!!
[image]
[image]
[image]

xuankuzcr, 2024-09-08:
stm32的GND还要和mid360的地线连到一起

wangdebin00223, 2024-09-09:
> stm32的GND还要和mid360的地线连到一起

我将mid360的地线和stm32连接到一起，还是报loss GPS signal Please check GPS signal 
并且我今天也更新了固件，不知道为什么会出现这个报错，一直不显示同步。
如果我按住stm32板的开关一直不放，就不会显示warning，会显示sampling但是还是显示 No Sync
请问您，有没有觉得哪里特别需要注意的点，我担心是我忽略了才导致的问题，但是我仔细排查了多遍，一直找不出问题所在
如果您能提供一些思路的话，十分感激！！！！

keepermzw, 2024-09-18:
我也遇到了同样的问题，但是我之前成功过，mid360的线剪短一次后就不行了，请问楼主解决了吗？

wangdebin00223, 2024-09-18:
> stm32的GND还要和mid360的地线连到一起

并没有解决，目前还没有排查出问题，换了线和雷达都不行

wangdebin00223, 2024-09-18:
> B站上有个博主有复现https://www.bilibili.com/video/BV14T421X7m9/?spm_id_from=333.788&vd_source=1291c9394eea1ba32ae2806fbb28c444

是的，我是按照他的方法来的，出现了以上问题。T-T

c6h6c6h6, 2024-09-19:
下周我买的器件也到了，准备复现看看。

xuankuzcr, 2024-09-20:
> > stm32的GND还要和mid360的地线连到一起
> 
> 我将mid360的地线和stm32连接到一起，还是报loss GPS signal Please check GPS signal 并且我今天也更新了固件，不知道为什么会出现这个报错，一直不显示同步。 如果我按住stm32板的开关一直不放，就不会显示warning，会显示sampling但是还是显示 No Sync 请问您，有没有觉得哪里特别需要注意的点，我担心是我忽略了才导致的问题，但是我仔细排查了多遍，一直找不出问题所在 如果您能提供一些思路的话，十分感激！！！！

我没遇到过这个问题，你检查下是不是虚焊了。如果后面解决了，也麻烦您在这个issue下同步下。

wangdebin00223, 2024-09-24:
> 我没遇到过这个问题，你检查下是不是虚焊了。如果后面解决了，也麻烦您在这个issue下同步下。
[image]
我用示波器测试了一下，它的波形好像有问题（GPS为绿色，PPS 为黄色），按理来说两个的频率应该都是1HZ但是有GPS的并不是，我更换了另一个STM32也是这个问题，一直找不到是哪里的问题

wangdebin00223, 2024-09-25:
> stm32的GND还要和mid360的地线连到一起
>
>
> 我将mid360的地线和stm32连接到一起，还是报loss GPS signal Please check GPS signal 并且我今天也更新了固件，不知道为什么会出现这个报错，一直不显示同步。 如果我按住stm32板的开关一直不放，就不会显示warning，会显示sampling但是还是显示 No Sync [...]

请问问作者有没有什么建议，供排查错误？我排查了很久找不到问题T-T

wangdebin00223, 2024-09-25:
> 我没遇到过这个问题，你检查下是不是虚焊了。如果后面解决了，也麻烦您在这个issue下同步下。

请问作者，这会不会是导致接收不到GPS信号的问题所在

qq1018408006, 2024-09-26:
> stm32的GND还要和mid360的地线连到一起
>
>
> 我将mid360的地线和stm32连接到一起，还是报loss GPS signal Please check GPS signal 并且我今天也更新了固件，不知道为什么会出现这个报错，一直不显示同步。 如果我按住stm32板的开关一直不放，就不会显示warning，会显示sampling但是还是显示 No Sync [...]

绿色的波形是GPRMC的串口信号，正点原子的示波器抓错频率很正常。从示波器的波形图上看没有啥问题。

你通过代码确认一下生成的GPRMC语句的最后一位的校验位是不是合法的，再用示波器抓一下mid360的GND和GPS输入两个引脚的波形看看和stm32的波形是不是一致（注意高低电平是不是一致）。看你串口波特率设置的是9600也没啥问题，我是想不到其他原因了。

上面提到的那个博主的复现方法我成功实现了，那个方法是没问题的。

另外搞这东西没点电子电路相关的基础知识是难受:(

wangdebin00223, 2024-09-26:
> 绿色的波形是GPRMC的串口信号，正点原子的示波器抓错频率很正常。从示波器的波形图上看没有啥问题。
>
> 你通过代码确认一下生成的GPRMC语句的最后一位的校验位是不是合法的，再用示波器抓一下mid360的GND和GPS输入两个引脚的波形看看和stm32的波形是不是一致（注意高低电平是不是一致）。看你串口波特率设置的是9600也没啥问题，我是想不到其他原因了。 [...]

请问，报文是不是合法的如何知道呀？以下是我输出的报文 应该是没有问题的
[image]
然后您说再用示波器抓一下mid360的GND和GPS输入两个引脚的波形看看和stm32的波形是不是一致，通过1分3的那条线，接到示波器进行检查吗？？
如果这些检查都没问题，不知道是不是雷达的问题，不知道有没有办法检查雷达

1466758326, 2024-09-26:
楼主，我也遇到的同样的问题，向您请教一下，雷达升级固件，这个意思是早期买的雷达需要升级雷达固件吗？方便提供一下链接吗？另外我也直接连接过gps板卡输出，依然是不行，您有合适的解决方法吗

1466758326, 2024-09-26:
> $GPRMC,000601.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2E $GPRMC,000602.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2D $GPRMC,000603.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2C $GPRMC,000604.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2B $GPRMC,000605.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2A $GPRMC,000606.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A29 $GPRMC,000607.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A28 $GPRMC,000608.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A27 $GPRMC,000609.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A*26 但是我将STM32的PA9和PB5分别接入MID360线的10号和8号线，在LivoxViewer查看Time Sync Type :显示 No Sync 并且有警告。能麻烦您帮忙分析一下，是哪里的问题吗？卡在这里好久了，一直排查不出来,十分感谢!!!! <img alt="072eec8038fc9cc30cefa93a17f01eb" width="335" src="https://private-user-images.githubusercontent.com/172002652/365390309-1f050dd4-84c4-4859-80b2-d3a0c4674946.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNDY2ODEsIm5iZiI6MTcyNzM0NjM4MSwicGF0aCI6Ii8xNzIwMDI2NTIvMzY1MzkwMzA5LTFmMDUwZGQ0LTg0YzQtNDg1OS04MGIyLWQzYTBjNDY3NDk0Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNlQxMDI2MjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZmM5NDViYWFlNjE4MzEyODE5ZjUyYzgzNWFlZjM3MTMwODg4MGJlYWE4MjQ1MWRjYTdjNGYxMTdlMWE3MDUxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ZOiU-plAGm1lyCajkAu0nYM889fX7vO17e5Cvi9qddo"> <img alt="07076ca5ebc5a2086bd16b3d5b2b703" width="584" src="https://private-user-images.githubusercontent.com/172002652/365390320-955fe25b-f647-4ecd-9b8f-138d874c80ba.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNDY2ODEsIm5iZiI6MTcyNzM0NjM4MSwicGF0aCI6Ii8xNzIwMDI2NTIvMzY1MzkwMzIwLTk1NWZlMjViLWY2NDctNGVjZC05YjhmLTEzOGQ4NzRjODBiYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNlQxMDI2MjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMzc1NzE4ZmUxOWJlODBkZDdjYmVlNjFlYTA5M2MyZjA4MmY4N2Q2N2ViYzFkYWU0ZmY2MzNiMTQ0MjJhMTY2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.1MqMtFeX7IxegyfWeY09FG58EmL9QE3eGGHGMViI9pg"> <img alt="53891c882ef7267103faa291aae27ba" width="477" src="https://private-user-images.githubusercontent.com/172002652/365390328-c1227500-4fd5-43d5-873b-fdc6c57f8a6d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNDY2ODEsIm5iZiI6MTcyNzM0NjM4MSwicGF0aCI6Ii8xNzIwMDI2NTIvMzY1MzkwMzI4LWMxMjI3NTAwLTRmZDUtNDNkNS04NzNiLWZkYzZjNTdmOGE2ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNlQxMDI2MjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YWVhYjMwZDg1ZThkNTQ4OTc5MGJlYzdjOWRjN2VhYjYyNzM1YjJmNmUwZDQ5MjY3MGRlYmViOWM1ZTgxNzUxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.s7hj5jtf0e9LmspTIQCuT4DFh0XE492gmR35wNqchgs">

我还想问一下，livox mid固件升级如何操作呢

wangdebin00223, 2024-09-26:
> 我还想问一下，livox mid固件升级如何操作呢

通过livox官网找到最新固件，然后可以看MID360手册或者网上搜一下LIVOX固件升级，有相应的教程。

wangdebin00223, 2024-09-26:
> 我还想问一下，livox mid固件升级如何操作呢

[image]

c6h6c6h6, 2024-09-27:
我调试通过了，同步成功。
方法就是用了B站的方法
stm32的代码发给你供参考

image.png (view on web)

1466758326, 2024-09-27:
刚才升级固件已经解决了，但是通过ros中的launch文件启动雷达之后，imu和雷达数据的时间戳仍然是本地的pc时间，和同步的时间不一样，这个如何解决呢？
[image]

wangdebin00223, 2024-09-27:
> 我调试通过了，同步成功。 方法就是用了B站的方法 stm32的代码发给你供参考
> […](#)

可能和我硬件本身有关系，您购买的新硬件应该问题不是出在代码上

wangdebin00223, 2024-09-27:
> 刚才升级固件已经解决了，但是通过ros中的launch文件启动雷达之后，imu和雷达数据的时间戳仍然是本地的pc时间，和同步的时间不一样，这个如何解决呢？ ![屏幕截图 2024-09-27 103033](https://private-user-images.githubusercontent.com/46621215/371371611-88750da6-2e5c-4d5c-abfc-b056521456a6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc0MDU5MTYsIm5iZiI6MTcyNzQwNTYxNiwicGF0aCI6Ii80NjYyMTIxNS8zNzEzNzE2MTEtODg3NTBkYTYtMmU1Yy00ZDVjLWFiZmMtYjA1NjUyMTQ1NmE2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI3VDAyNTMzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgzOGFmNzE4NTFkNzA0NDk1ZjA4YjMzYzc5NjQwZWE5NTA4MTRlZDBhNzg1OGFkYTZmMTA1MzIyNTdkNTU0ODMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.yfH0SO5zfAVMXuJtvdYvcczsNcMTVNhwJNUFThqbMd0)

没太理解，我还没做到这里...可以咨询下博主

1466758326, 2024-09-27:
> 没太理解，我还没做到这里...可以咨询下博主

这个意思是这样的，我是用板卡，模拟了一个和pc不一样的过去时间，比如2023年1月1日的时间，但是ros的驱动启动之后，发出的imu和雷达消息的时间戳仍然是本地pc时间，并不是gps同步的时间

wangdebin00223, 2024-09-27:
> 这个意思是这样的，我是用板卡，模拟了一个和pc不一样的过去时间，比如2023年1月1日的时间，但是ros的驱动启动之后，发出的imu和雷达消息的时间戳仍然是本地pc时间，并不是gps同步的时间

可以留下联系方式一起交流帮助，在这上面比较慢

1466758326, 2024-09-27:
> 可以留下联系方式一起交流帮助，在这上面比较慢

好的呀，v:13837578204

c6h6c6h6, 2024-09-27:
我准备在接收端程序内设置一个全局变量，接收到第一帧时存储当前时间和第一帧的时间差——delta_t，后续的每一帧还原时间都是帧原时间加上delta_t。

> 没太理解，我还没做到这里...可以咨询下博主

1466758326, 2024-09-27:
> 我准备在接收端程序内设置一个全局变量，接收到第一帧时存储当前时间和第一帧的时间差——delta_t，后续的每一帧还原时间都是帧原时间加上delta_t。

这个意思是，修改ros drvier程序？？

wangdebin00223, 2024-10-03:
> 我没遇到过这个问题，你检查下是不是虚焊了。如果后面解决了，也麻烦您在这个issue下同步下。

您好作者，经过多方面的排查，发现是所使用的MID360 接口的模组损坏，现以保修。折腾了好久，感谢之前您提供的建议

xuankuzcr, 2024-10-05:
不客气，解决问题就好。GitHub上的同步方案经过多次修正已经没啥bug了，如果有问题一般是硬件问题。

wangdebin00223, 2024-10-07:
> 不客气，解决问题就好。GitHub上的同步方案经过多次修正已经没啥bug了，如果有问题一般是硬件问题。

好的，恭喜您的论文被顶刊收录！！

closed 2024-11-10 by wangdebin00223

keepermzw, 2026-01-31:
换了一个雷达发现可以。寄回检修后客服告知接口模组损坏。

怎么排查出 MID360 接口的模组损坏的呢？目前遇到了同样的问题