hku-mars/FAST-LIVO #128: 您好打扰了，我在运行示例数据时rviz无法正常工作，且存在IMUProcess的WARN

XIONG-Jun-dev, 2024-09-19:
所有的示例数据（hku1、hku2等）都试过了，在运行时有两个问题：
1. 启动时会有告警，并提示FAST-LIVO not ready，请问是正常的吗，如果不是应该如何解决？
[image]

2. 运行后rviz无法正常显示点云效果，只显示轨迹
[image]
卡了好几天了，重装了几次系统还是没解决，一直找不到原因，十分感谢您的解答！！

xuankuzcr, 2024-09-20:
参考下issue https://github.com/hku-mars/FAST-LIVO/issues/53
看看能不能解决，如果仍然有问题的话，在这个issue里继续同步一下

XIONG-Jun-dev, 2024-09-21:
> 参考下issue #53 看看能不能解决，如果仍然有问题的话，在这个issue里继续同步一下

感谢回复！！我根据您提供的参考链接进行了重装，两个问题仍然存在。我把环境配置过程中的一些报错贴出来，您看有可能是这些报错导致的么？
1. catkin_make livox驱动过程中：
[image]

2. catkin_make fastlivo项目时：
[image]
[image]

xuankuzcr, 2024-10-03:
看起来编译过程没什么问题，可以查一下你的Eigen版本号嘛

XIONG-Jun-dev, 2024-10-08:
> 看起来编译过程没什么问题，可以查一下你的Eigen版本号嘛

感谢回复！我的Eigen版本号是3.3.7，您看需要更新到哪个指定版本么？

xuankuzcr, 2024-10-11:
看你的耗时部分和左下的tracked image，视觉和LiDAR部分都是在正常运行的，你可以rostopic echo下/cloud_registered这个话题看看有没有数据

pankan2, 2025-01-06:
> 看你的耗时部分和左下的tracked image，视觉和LiDAR部分都是在正常运行的，你可以rostopic echo下/cloud_registered这个话题看看有没有数据

[image]
打扰了 我运行后他没数据呢 可能是什么问题呢？

stupidnb222, 2025-04-06:
您好，我这边也是这种情况，我对cloud_registered 这个话题监控了一下，是有数据的，并且好像也附带rgb数值，但是rviz也还是没有显示彩色地图，也尝试过更改avia.config 文件[](url)

[image]

stupidnb222, 2025-04-06:
我是使用的虚拟机，有关系吗？

stupidnb222, 2025-04-07:
解决了，pointcloud2 里面decay time 设置一下就好了，我是笨蛋=/=