hku-mars/FAST-LIVO #131: double free or corruption (out)

BloodLemonS, 2024-09-27:
How to solve the problem of double free or corruption (out) after running?

xuankuzcr, 2024-10-03:
Could you provide more information?

BloodLemonS, 2024-10-09:
The running result is shown in the following figure
x-special/nautilus-clipboard
copy
file:///home/hp/Desktop/2024-10-09%2013-41-32%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE.png
x-special/nautilus-clipboard
copy
file:///home/hp/Desktop/2024-10-09%2013-41-16%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE.png

BloodLemonS, 2024-10-09:
The running result is shown in the following figure.
[image]
[image]

BloodLemonS, 2025-01-02:
[image]
通过debug模式下，目前也没有看出问题出现在什么地方。是内存不够用吗？我是16G内存大小