hku-mars/FAST-LIVO #133: Sophus 2.3  在编译时不通过（已解决）

woshizhangbowen, 2024-10-10:
zbw@zbw-virtual-machine:/media/zbw/zbw2/zbw/fast-livo/Sophus/build$ make
/usr/local/bin/cmake -S/media/zbw/zbw2/zbw/fast-livo/Sophus -B/media/zbw/zbw2/zbw/fast-livo/Sophus/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/bin/cmake -E cmake_progress_start /media/zbw/zbw2/zbw/fast-livo/Sophus/build/CMakeFiles /media/zbw/zbw2/zbw/fast-livo/Sophus/build//CMakeFiles/progress.marks
make  -f CMakeFiles/Makefile2 all
make[1]: 进入目录“/media/zbw/zbw2/zbw/fast-livo/Sophus/build”
make  -f CMakeFiles/Sophus.dir/build.make CMakeFiles/Sophus.dir/depend
make[2]: 进入目录“/media/zbw/zbw2/zbw/fast-livo/Sophus/build”
cd /media/zbw/zbw2/zbw/fast-livo/Sophus/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /media/zbw/zbw2/zbw/fast-livo/Sophus /media/zbw/zbw2/zbw/fast-livo/Sophus /media/zbw/zbw2/zbw/fast-livo/Sophus/build /media/zbw/zbw2/zbw/fast-livo/Sophus/build /media/zbw/zbw2/zbw/fast-livo/Sophus/build/CMakeFiles/Sophus.dir/DependInfo.cmake --color=
make[2]: 离开目录“/media/zbw/zbw2/zbw/fast-livo/Sophus/build”
make  -f CMakeFiles/Sophus.dir/build.make CMakeFiles/Sophus.dir/build
make[2]: 进入目录“/media/zbw/zbw2/zbw/fast-livo/Sophus/build”
[  5%] Building CXX object CMakeFiles/Sophus.dir/sophus/so2.cpp.o
/usr/bin/c++ -DSophus_EXPORTS -I/usr/local/include/eigen3 -O3 -DNDEBUG  -fPIC   -Wall -Werror -Wno-unused-variable                    -Wno-unused-but-set-variable -Wno-unknown-pragmas  -MD -MT CMakeFiles/Sophus.dir/sophus/so2.cpp.o -MF CMakeFiles/Sophus.dir/sophus/so2.cpp.o.d -o CMakeFiles/Sophus.dir/sophus/so2.cpp.o -c /media/zbw/zbw2/zbw/fast-livo/Sophus/sophus/so2.cpp
/media/zbw/zbw2/zbw/fast-livo/Sophus/sophus/so2.cpp: In constructor ‘Sophus::SO2::SO2()’:
/media/zbw/zbw2/zbw/fast-livo/Sophus/sophus/so2.cpp:32:26: error: lvalue required as left operand of assignment
   32 |   unit_complex_.real() = 1.;
      |                          ^~
/media/zbw/zbw2/zbw/fast-livo/Sophus/sophus/so2.cpp:33:26: error: lvalue required as left operand of assignment
   33 |   unit_complex_.imag() = 0.;
      |                          ^~
make[2]: *** [CMakeFiles/Sophus.dir/build.make:79：CMakeFiles/Sophus.dir/sophus/so2.cpp.o] 错误 1
make[2]: 离开目录“/media/zbw/zbw2/zbw/fast-livo/Sophus/build”
make[1]: *** [CMakeFiles/Makefile2:98：CMakeFiles/Sophus.dir/all] 错误 2
make[1]: 离开目录“/media/zbw/zbw2/zbw/fast-livo/Sophus/build”
make: *** [Makefile:149：all] 错误 2

在so2.cpp下面第30行  修改为以下

SO2::SO2()
{
  // unit_complex_.real() = 1.;
  // unit_complex_.imag() = 0.;
  unit_complex_ = std::complex<double>(1.0, 0.0);
}

ok

xuankuzcr, 2024-10-11:
这是非模板类Sophus比较经典的报错了，感谢指出。其他问题可以参考以前的issue，都能搜到解决方案。

whyscience, 2024-12-04:
https://blog.csdn.net/Son_of_the_Bronx/article/details/138212308

SO2::SO2()
{
  unit_complex_.real(1.);
  unit_complex_.imag(0.);
}