hku-mars/FAST-LIVO #134: 相机视野之外的点云被上色

qq1018408006, 2024-10-10:
[image]
我尝试使用mid360代替avia来复现这个工作，在雷达内外参都标定好的情况下，在建图时总会有相机视野之外的点云被上色并且参与建图（绿色线之间的点云），从俯视图可以明显地看出真实的相机视场角小于被上色的点云的视场角。

有没有人遇到同样的问题？

xuankuzcr, 2024-10-11:
这是背后点云投影到相机成像平面上了，已修复 (见commit 720f5d828993be639fe0c34e23d221423149e87f)。

KLMN2001, 2024-11-05:
遇见了相同的问题，我使用的是robosence HELIOS_16P代替avia，由于雷达扫描范围为360度，最初测试时存在相机背面仍存在点云且被上色，按照commit 720f5d8修改后背面点云已经被剔除，但与上面问题类似，俯视图下在相机视野之外左右各有一块区域存在被上色点云，且左右区域与中间存在很明显的一段空白区域，看起来很奇怪。
不知道您是否解决了该问题？
[image]

qq1018408006, 2024-11-11:
@KLMN2001 已经解决。是fast livo采用了cv::projectPoints函数导致的问题，这个函数在存在相机成像畸变时可能会将视野之外的点投影到带畸变的图像中去，从而导致视野之外的点云被上色。我已经将cv::projectPoints替换为手动图像去畸变+点云投影到未畸变的图像上的代码，虽然数据处理速度变慢了但是解决了问题。

KLMN2001, 2024-11-12:
感谢您的回复，我这边参照 commit 720f5d8 的方式将FOV外的点云剔除，虽然没找到原因所在但也算是解决了问题，下面的代码供您参考
fast_livo/src/FAST-LIVO/src/laserMapping.cpp/void publish_frame_world_rgb(const ros::Publisher & pubLaserCloudFullRes, lidar_selection::LidarSelectorPtr lidar_selector)

        // 定义水平FOV范围
        const double horizontal_fov_deg = 72;
        const double half_horizontal_fov_rad = (horizontal_fov_deg / 2.0) * M_PI / 180.0;

            // 计算水平角度
            double azimuth_angle_rad = atan2(pf[0], pf[2]);
            // 检查水平角度是否在FOV范围内
            if (azimuth_angle_rad < -half_horizontal_fov_rad || azimuth_angle_rad > half_horizontal_fov_rad) continue;

xuankuzcr, 2024-11-12:
感谢 @qq1018408006 明确指出了`projectPoints`存在的问题。
感谢 @KLMN2001 检查水平FoV的方法既保证了效率，又避免了视野之外的点云被上色。