hku-mars/FAST-LIVO #135: ROS free installation

rohitashwin, 2024-10-17:
Thanks for this project!

I want to use this without ROS. What are the modifications I need to make in order to make it work without ROS?

rohitashwin, 2024-10-17:
I don't need support for Livox

xuankuzcr, 2024-10-20:
Here are some suggestions for you! 😊
1. Replace ros::spin() with standard C++ threads or callbacks; 
2. Use sensor SDKs directly (without Livox); 
3. Use Pangolin or OpenGL to replace rviz for visualization.

MGforest, 2024-10-22:
Use Docker #136 
You need to git clone the repo on your local machine, but don't need to compile it locally.

xuankuzcr, 2024-10-26:
You're welcome! Feel free to reach out if you encounter any challenges. Regarding your question, Livox is required because the dataset relies on its message format.