hku-mars/FAST-LIVO #136: Use with Docker

MGforest, 2024-10-22:
Hi, 
I created a Dockerfile for it. Save it as "Dockerfile", without file extension.

```
FROM osrf/ros:noetic-desktop-full

# Set non-interactive mode to avoid prompts
ENV DEBIAN_FRONTEND=noninteractive

# Install necessary system dependencies and ROS tools (excluding pre-built PCL ROS packages)
RUN apt-get update && apt-get install -y \
    gedit \
    nano \
    git \
    build-essential \
    cmake \
    python3-pip \
    python3-rosdep \
    python3-catkin-tools \
    locales \
    libeigen3-dev \
    libceres-dev \
    libopencv-dev \
    libyaml-cpp-dev \
    ros-noetic-cv-bridge \
    ros-noetic-image-transport \
    ros-noetic-tf2-geometry-msgs \
    ros-noetic-message-filters \
    ros-noetic-geometry-msgs \
    ros-noetic-sensor-msgs \
    ros-noetic-eigen-conversions \
    ros-noetic-catkin \
    && rm -rf /var/lib/apt/lists/*

# Install PCL 1.13 from source
RUN mkdir -p /opt/pcl && \
    cd /opt/pcl && \
    git clone https://github.com/PointCloudLibrary/pcl.git && \
    cd pcl && \
    git checkout pcl-1.13.0 && \
    mkdir build && \
    cd build && \
    cmake .. -DCMAKE_BUILD_TYPE=Release && \
    make -j"$(nproc)" && \
    make install && \
    ldconfig

# Install Sophus (with fix for so2.cpp)
RUN cd /opt && \
    rm -rf Sophus && \
    git clone https://github.com/strasdat/Sophus.git && \
    cd Sophus && \
    git checkout a621ff && \
    sed -i 's/unit_complex_.real() = 1\.;/unit_complex_ = std::complex<double>(1,0);/' sophus/so2.cpp && \
    sed -i 's/unit_complex_.imag() = 0\.;/unit_complex_ = std::complex<double>(1,0);/' sophus/so2.cpp && \
    mkdir build && cd build && \
    cmake .. && make -j"$(nproc)" && make install

# Create the workspace and build Fast-LIVO in it
ENV WORKSPACE=/catkin_ws

RUN mkdir -p $WORKSPACE/src

# Clone Vikit
RUN cd $WORKSPACE/src && \
    git clone https://github.com/uzh-rpg/rpg_vikit.git && \
    cd rpg_vikit/vikit_common/src && \
    sed -i '48s/CV_RANSAC/cv::RANSAC/' homography.cpp && \
    sed -i '112s/CV_INTER_LINEAR/cv::INTER_LINEAR/' pinhole_camera.cpp && \
    sed -i '237s/CV_WINDOW_AUTOSIZE/cv::WINDOW_AUTOSIZE/' img_align.cpp && \
    sed -i '437s/CV_WINDOW_AUTOSIZE/cv::WINDOW_AUTOSIZE/' img_align.cpp

# Clone and build livox_ros_driver
RUN cd $WORKSPACE/src && \
    git clone https://github.com/Livox-SDK/livox_ros_driver.git

# Clone and build Fast-LIVO
RUN cd $WORKSPACE/src && \
    git clone https://github.com/hku-mars/Fast-LIVO.git && \
    cd Fast-LIVO && \
    git submodule update --init --recursive

    
RUN cd $WORKSPACE && \
    /bin/bash -c "source /opt/ros/noetic/setup.bash && catkin_make"

WORKDIR /catkin_ws

```

then go to your local Fast-LIVO folder. I created a "data" folder. 
PWD : ~/catkin_ws/src/FAST-LIVO/data
The dot says, the Dockerfile is located in the current folder, can be anywhere, but I have in the Fast-LIVO folder. 

`$ docker build -t fast-livo .`

`$docker run -it -e "DISPLAY" -e "QT_X11_NO_MITSHM=1"     -v "/tmp/.X11-unix:/tmp/.X11-unix:rw"     -v "$PWD:/data" fast-livo`

in the container: 
`/catkin_ws# source devel/setup.bash`

`/catkin_ws# roslaunch fast_livo mapping_avia.launch`

add a new terminal: 
in a new local terminal, type:
`$ docker ps`
`$ docker exec -it -e DISPLAY -e QT_X11_NO_MITSHM=1 CONTAINTER-ID bash`
`/catkin_ws# rosbag play /data/lidar_cam.bag`

This was based on #53 and https://github.com/ethz-asl/kalibr/wiki/installation

closed 2024-10-22 by MGforest

BloodLemonS, 2025-01-06:
Thank you for providing the Docker configuration method, which is very effective. May I ask what is the OpenCV version configured in your Docker? I am using OpenCV version 3.4.15, and I encounter the following problem while running. How can I solve it?
OpenCV(3.4.15) /root/opencv/modules/core/src/matrix.cpp:224: error: (-215:Assertion failed) 0 <= _dims && _dims <= CV_MAX_DIM in function 'setSize'
[laserMapping-2] process has died [pid 128, exit code -6, cmd /fastlivo_ws/devel/lib/fast_livo/fastlivo_mapping __name:=laserMapping __log:=/root/.ros/log/e7a58190-cc02-11ef-a31c-0242ac110002/laserMapping-2.log].
log file: /root/.ros/log/e7a58190-cc02-11ef-a31c-0242ac110002/laserMapping-2*.log

MGforest, 2025-01-09:
Sorry for the late reply. 
OpenCV Version 4.2.0 (Version: 4.2.0+dfsg-5+focal3)

It would be better to specify it in the Dockerfile, sorry.