hku-mars/FAST-LIVO #139: 关于时间同步后，32时间赋值的问题，想请教一下

Hohaizzz, 2024-10-30:
十分感谢作者的开源工作，同时十分期待livo2的精彩表现。
这几天跟新了海康相机型号，可以使用stm32进行成功的时间同步，跑出来的效果也交以往未同步时稳定很多，但是还需将一个红外相机一起录制bag，因为这个红外相机无外部触发源，只有一个type-c口。因此同步后的雷达和海康相机的时间戳是15开头的，2020年的，红外相机则是目前时间。想请教一下，怎么去修改，让32触发的时间是当前时间呢，使用ntp时间校准可行吗。十分感谢！！
[image]

Hohaizzz, 2024-10-30:
[image]

xuankuzcr, 2024-10-31:
感谢您的支持！我觉得你可以在接收到GPRMC的同时记录下此时的系统时间，这两者的bias就是32时钟和系统时钟的time offset。以下是livox_ros_driver中的代码片段，可以将其改为这种形式来计算time offset。
```cpp
#include <ros/ros.h>
#include <ctime>

// 解析 $GPRMC
while (kParseSuccess == comm_->ParseCommStream(&packet)) {
    if ((fn_cb_ != nullptr) && 
        ((strstr((const char *)packet.data, "$GPRMC")) || 
         (strstr((const char *)packet.data, "$GNRMC")))) {
        
        // 提取 GPRMC 的时间和日期字段
        std::string gprmc_data((const char *)packet.data);
        int hours = std::stoi(gprmc_data.substr(7, 2));
        int minutes = std::stoi(gprmc_data.substr(9, 2));
        int seconds = std::stoi(gprmc_data.substr(11, 2));
        int day = std::stoi(gprmc_data.substr(gprmc_data.find(",", 9) + 1, 2));
        int month = std::stoi(gprmc_data.substr(gprmc_data.find(",", 9) + 3, 2));
        int year = 2000 + std::stoi(gprmc_data.substr(gprmc_data.find(",", 9) + 5, 2));

        // 设置 tm 结构直接生成 GPRMC 时间戳
        std::tm gprmc_tm = {seconds, minutes, hours, day, month - 1, year - 1900};
        std::time_t gprmc_timestamp = std::mktime(&gprmc_tm);

        // 获取当前系统（ROS）时间戳
        std::time_t ros_timestamp = ros::Time::now().sec;

        // 计算时间偏差
        double time_offset = difftime(ros_timestamp, gprmc_timestamp);
        printf("Time offset between GPRMC and ROS time: %.2f seconds\n", time_offset);

        // 调用回调函数
        fn_cb_((const char *)packet.data, packet.data_len, client_data_);
        
        // 输出解析成功消息
        printf("RMC data parse success!\n");
    }
}
```

closed 2024-10-31 by Hohaizzz