hku-mars/FAST-LIVO #140: 用rgbd相机可以跑吗? 比如D435i

yongming-foxx, 2024-10-30:


xuankuzcr, 2024-10-31:
> 如题. 有人测试过吗? fast-livo是不是依赖于lidar这种精确的距离传感器呢?

理论上可以用rgbd，但目前我没做过这方面的开发，但起码L515应该是可以跑的

thudongfeng, 2024-10-31:
首先，FAST-LIVO要求的时间同步是需要相机、雷达、imu的时间信息统一到一个时间上，如2023年08分03秒369毫秒
其次，对激光雷达进行时间同步一般是通过获取GPS时间进行同步，这里推荐采用论文课题组所提出的采用STM32模拟授时设备的方式，经过同步后雷达输出的电源信息就是STM32所发送的时间，因为LIVOX雷达内部集成了IMU，所以IMU的时间会自动对齐到雷达，也就是GPS时间信息
最后，相机的时间同步目前是推荐采用海康的相机，能够实现用STM32进行外部触发，在触发后通过访问雷达所共享出的时间信息进行对齐到GPS上
综上，如果想用D435i相机复现FAST-LIVO需要自行魔改相机驱动，在STM32信号触发深度相机拍摄后，通过读取共享内存中的雷达时间去作为RGB图像的时间，这样才能实现RGB信息和雷达、imu的同步
此外推荐[LIV_handhold](https://github.com/sheng00125/LIV_handhold)去进行FAST-LIVO算法的复现

Jack-XHP, 2024-10-31:
I have a question what if instead using a lidar, depth output is projected into point cloud to replace lidar point cloud. Then using a d435i rgb, depth and IMU can be synced. Will FAST-LIVO be able to work with such setup?

yongming-foxx, 2024-10-31:
> I have a question what if instead using a lidar, depth output is projected into point cloud to replace lidar point cloud. Then using a d435i rgb, depth and IMU can be synced. Will FAST-LIVO be able to work with such setup?

I don't fully understand your question. Currently I'm unsure about the exact timing requirements for synchronization—whether it’s around 1 ms or 10 ms, for example.

closed 2024-11-21 by yongming-foxx