hku-mars/FAST-LIVO #141: 海康21UC相机，MVS 设置 trigger on 失败

JACKLiuDay, 2024-11-04:
郑博您好，我在使用海康的21UC的过程中，在MVS中设置trigger mode 为 on，但是获取不到图像， trigger mode 改成 off 就可以立马获取到图像了，这是什么原因造成的呢。MID360雷达倒是同步成功了显示 GPS sync。

JACKLiuDay, 2024-11-04:
> 郑博您好，我在使用海康的21UC的过程中，在MVS中设置trigger mode 为 on，但是获取不到图像， trigger mode 改成 off 就可以立马获取到图像了，这是什么原因造成的呢。MID360雷达倒是同步成功了显示 GPS sync。

发现问题了，相机的地线在32上接成6号线了，应该接5号线

closed 2024-11-04 by JACKLiuDay

xuankuzcr, 2024-11-04:
是的，应该连OPTO_GND，光耦隔离的那个地，readme里面有写。我在config里设置的trigger mode是为了在无stm32时测试相机用的。
[image]

JACKLiuDay, 2024-11-04:
> 是的，应该连OPTO_GND，光耦隔离的那个地，readme里面有写。我在config里设置的trigger mode是为了在无stm32时测试相机用的。 [...]

感谢大佬回复，郑博最近有开源FAST-LIVO2的打算吗，等三个月了:rofl: