hku-mars/FAST-LIVO #142: 海康相机时间戳 173开头，mid360雷达时间戳159开头

JACKLiuDay, 2024-11-04:
郑博您好，我硬件触发都正常，但是看着21UC相机的数据时间戳和雷达点云的时间戳不一致，雷达的是159开头，相机是173开头，这是什么原因啊

xuankuzcr, 2024-11-04:
LiDAR是stm32给的GPRMC时间，相机是系统时间，看起来是相机没读到共享内存里的时间戳，然后打的ros::Time::now()。如果用的都是仓库里面的driver，不应该有这个问题。检查下/home/usr_name/下有没有timeshare这个文件。

JACKLiuDay, 2024-11-04:
> LiDAR是stm32给的GPRMC时间，相机是系统时间，看起来是相机没读到共享内存里的时间戳，然后打的ros::Time::now()。如果用的都是仓库里面的driver，不应该有这个问题。检查下/home/usr_name/下有没有自动创建timeshare这个文件。

home/user/下面没有timeshare文件夹啊

JACKLiuDay, 2024-11-05:
> > LiDAR是stm32给的GPRMC时间，相机是系统时间，看起来是相机没读到共享内存里的时间戳，然后打的ros::Time::now()。如果用的都是仓库里面的driver，不应该有这个问题。检查下/home/usr_name/下有没有自动创建timeshare这个文件。
> 
> home/user/下面没有timeshare文件夹啊

用仓库里提供的driver就可以了，我之前用的官网上下载的 livox driver

closed 2024-11-05 by JACKLiuDay

pankan2, 2025-01-06:
用仓库里提供的driver就可以了，我之前用的官网上下载的 livox driver 请问一下这个是什么意思啊？