hku-mars/FAST-LIVO #158: 关于使用其他数据集的配置问题

DioBakery, 2024-12-01:
使用私有数据集的时候能够很好地运行，但是使用其他数据集时候不确定要设定哪些参数，导致可能在开启img_enable时效果不如关闭时的效果，甚至产生较大漂移
例如：使用[m2dgr](https://github.com/SJTU-ViSYS/M2DGR)的gate01 bag时，使用如下参数：
config.yaml
> feature_extract_enable : 0
> point_filter_num : 2
> max_iteration : 10
> dense_map_enable : 1
> filter_size_surf : 0.15
> filter_size_map : 0.15
> cube_side_length : 20
> grid_size : 40
> patch_size : 8
> img_enable : 1
> lidar_enable : 1
> outlier_threshold : 300 # 78 100 156
> ncc_en: false
> ncc_thre: 0
> img_point_cov : 1000 # 1000
> laser_point_cov : 0.001 # 0.001
> pose_output_en: false
> delta_time: 0.0
> 
> common:
>     lid_topic:  "/velodyne_points"
>     imu_topic:  "/handsfree/imu"
> 
> preprocess:
>     lidar_type: 2 # 1:Livox LiDAR  2:VELO16  3:OUST64  4:XT32  5:Mid360
>     scan_line: 32 # 16  64  32
>     blind: 0.5 # blind x m disable
> 
> mapping:
>     acc_cov_scale: 100
>     gyr_cov_scale: 10000
>     extrinsic_T: [0.27255, -0.00053,0.17954] # lidar to imu
>     extrinsic_R: [1, 0, 0,
>                   0, 1, 0,
>                   0, 0, 1]
> 
> pcd_save:
>     pcd_save_en: false
> 
> camera:
>     img_topic: /camera/color/image_raw
>     Rcl: [0, -1, 0,
>           0, 0, -1,
>           1, 0, 0]
>     Pcl: [-0.27255, 0.00053, -0.17954]

camera.yaml

> cam_model: Pinhole
> cam_width: 640
> cam_height: 480
> cam_fx: 617.971050917033
> cam_fy: 616.445131524790
> cam_cx: 327.710279392468
> cam_cy: 253.976983707814
> cam_d0: 0.148000794688248
> cam_d1: -0.217835187249065
> cam_d2: 0
> cam_d3: 0
效果如下：
**img_enable=1：**
[image]
**img_enable=0：**
[image]
相机和雷达的内外参是根据数据集标定配置的，但其他参数不太确定作用，烦请指教。

xuankuzcr, 2024-12-01:
m2dgr相机和LiDAR的数据未同步，从你第一张图的残影就可以看出来。由于相机和LiDAR的时间戳不是他们真实采样时刻的时间戳，不同时空系的数据加入一定会产生负优化。可以通过调整`delta_time`参数手调出一个time offset。

DioBakery, 2024-12-02:
> m2dgr相机和LiDAR的数据未同步，从你第一张图的残影就可以看出来。由于相机和LiDAR的时间戳不是他们真实采样时刻的时间戳，不同时空系的数据加入一定会产生负优化。可以通过调整`delta_time`参数手调出一个time offset。

[了解了，但是如果camera是15hz而lidar是10hz，所以尝试[S3E](https://pengyu-team.github.io/S3E/)数据集。
但发现p_imu->Process2(LidarMeasures, state, feats_undistort);这一步始终feats_undistort的size为0，使得一直无法建立ikdforest树，请问这是怎么回事呢？
```
#ifdef USE_IKFOM
        p_imu->Process(LidarMeasures, kf, feats_undistort);
        state_point = kf.get_x();
        pos_lid = state_point.pos + state_point.rot * state_point.offset_T_L_I;
        #else
        p_imu->Process2(LidarMeasures, state, feats_undistort);
        state_propagat = state;
        #endif

        if (lidar_selector->debug)
        {
            LidarMeasures.debug_show();
        }
        std::cout << "LidarMeasures.lidar->points.size(): " << LidarMeasures.lidar->points.size() << std::endl;
        std::cout << "feats_undistort->points.size(): " << feats_undistort->points.size() << std::endl;
```
output:
> LidarMeasures.lidar->points.size(): 26583
> feats_undistort->points.size(): 0