hku-mars/FAST-LIVO #164: 为什么我的点云没有显示

pankan2, 2025-01-06:
已经输出了 点云和图像 但是在rviz中没点云显示呢 

[ LIO ]: Raw feature num: 4607 downsamp num 1603 Map num: 2693.
[ LIO ]: Using multi-processor, used core number: 4.
[ LIO ]: time: fov_check: 0.000000 fov_check and readd: 0.000850 match: 0.005765 solve: 0.000380  ICP: 0.003161  map incre: 0.000663 total: 0.009576 icp: 0.007371 construct H: 0.000109.
[ INFO ]: get point cloud at time: 1590192960.544835.
feature extraction time: 0.000231 
[ INFO ]: get img at time: 1736141493.127414.
[ LIO ]: Raw feature num: 4501 downsamp num 1626 Map num: 2691.
[ LIO ]: Using multi-processor, used core number: 4.
[ LIO ]: time: fov_check: 0.000000 fov_check and readd: 0.000961 match: 0.005761 solve: 0.000380  ICP: 0.003620  map incre: 0.000642 total: 0.009569 icp: 0.007366 construct H: 0.000109.
[ INFO ]: get img at time: 1736141493.127414.
[ INFO ]: get point cloud at time: 1590192960.644675.
feature extraction time: 0.000199 
[ INFO ]: get img at time: 1736141493.227597.
[ LIO ]: Raw feature num: 4639 downsamp num 1621 Map num: 2694.
[ LIO ]: Using multi-processor, used core number: 4.
[ LIO ]: time: fov_check: 0.000000 fov_check and readd: 0.001101 match: 0.005760 solve: 0.000380  ICP: 0.006571  map incre: 0.000594 total: 0.009567 icp: 0.007364 construct H: 0.000109.
[ INFO ]: get img at time: 1736141493.227597.
[ INFO ]: get point cloud at time: 1590192960.744514.

[image]

highhunter777, 2025-01-11:
i have same problem too.

dzyzyy, 2025-01-19:
我现在也是，driver2用的是郑博的，海康用的是官方的，起初不接stm32还可以触发vio，后面接上单片机stm32，直接又不触发了，断了stm32也不触发了。过了几天重新打开电脑不接stm32静止又能触发。

[image]

dzyzyy, 2025-01-21:
别用海康官方ros封装，用郑博的才能触发vio。

yubo105139, 2025-02-17:
@pankan2 有结果了吗，我也是，只有点，无法建图

yubo105139, 2025-02-18:
@dzyzyy 您好，您解决了吗

Bio-Neal, 2025-04-15:
我怎么图像也没有，只有轨迹