hku-mars/FAST-LIVO #169: 为什么 我的点云保存失败呢

pankan2, 2025-01-20:
我的pcd_save_en为true 
运行fast_livo 在rviz中显示点云 ，然后在ctrl+c关闭fast_livo 
[image]

xuankuzcr, 2025-01-20:
拉一下最新代码再试一下，看你的参数文件代码版本很老了。

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2025年01月20日 18:00 |
| 收件人 | hku-mars/FAST-LIVO ***@***.***> |
| 抄送人 | Subscribed ***@***.***> |
| 主题 | [hku-mars/FAST-LIVO] 为什么 我的点云保存失败呢 (Issue #169) |

我的pcd_save_en为true
运行fast_livo 在rviz中显示点云 ，然后在ctrl+c关闭fast_livo
2025-01-20.17-45-35.png (view on web)

pankan2, 2025-01-21:
> 拉一下最新代码再试一下，看你的参数文件代码版本很老了。
> […](#)
> 
> ---- 回复的原邮件 ----
> |发件人 |***@***.***> |
> |发送日期 |2025年01月20日 18:00 |
> |收件人 |hku-mars/FAST-LIVO ***@***.***> |
> |抄送人 |订阅 ***@***.***> |
> |主题 |[hku-mars/FAST-LIVO] 为什么 我的点云保存失败呢 （Issue[#169](https://github.com/hku-mars/FAST-LIVO/issues/169)) |
> 
> 我的pcd_save_en为true
> 运行fast_livo 在 rviz中显示点云 ，然后在ctrl+c关闭fast_livo
> 2025-01-20.17-45-35.png （在 Web 上查看）
> 
> —
> 直接回复此电子邮件，在 GitHub 上查看，或取消订阅。
> 您收到此消息是因为您订阅了此线程。消息 ID：***@***.***>

大佬好 fast_livo我下载的这个不是最新的吗？ 昨天我问另一位大佬说可能是因为我的机器性能太弱，ctrl+C触发后，线程都来不及操作，进程都超时终止了，我的fast_livo在rviz中很卡，运行到后面几乎是1帧/0帧， 我使用的是N100 大佬

[image]

pankan2, 2025-01-21:
谢谢郑博 在这里有解决方案
https://github.com/hku-mars/FAST-LIVO/issues/94#issuecomment-2141143870

closed 2025-01-21 by pankan2