hku-mars/FAST-LIVO #173: [laserMapping-1]/[laserMapping-2] process has died/Double Free

prexhu, 2025-03-04:
# SysInfo
OS: Ubuntu 20.04.6 LTS x86_64
Kernel: 5.15.0-131-generic
OpenCV 4.2.0

感谢郑博以及Mars-Lab成员的开源这么棒的工作，刚刚也测试了FAST-LIVO2，效果更加惊艳！祝好：）

这个issue主要是我个人在跑通FAST-LIVO* Datasets的时候解决类似#131 #143  #154 #41 #31 出现的问题

先说结论：重装库[pcl1.14](https://github.com/PointCloudLibrary/pcl/releases/tag/pcl-1.14.0)
听起来确实是很玄学，我第一出现的问题是 `OpenCV` #143 #154 报出的错误，查看了相关的issue，但是并没有解决我的问题，直到我重装了`pcl-1.14`(在下载的源码里先`sudo make uninstall` 然后再次`sudo make install`)。

第二次出现的问题同样是`OpenCV`报出的错误，我像第一次一样重装了`pcl-1.14`。`OpenCV`的错误没有了，但是紧接而来的就是`double free or corruption (out)` #131，要么是直接报错，要么是运行一段时间之后报出错误，我查看了相关issue， #31 中有解决方案，但是对我代价太大了，卡了好几天，这其中我使用了`launch prefix`来debug，gdb给出的信息都和以上的issue给出的类似，都是很底层的问题，今天我无意间解决了这个问题，先卸载`pcl-1.14`,然后`catkin_make`一下工作空间，等它报错，然后再次重装`pcl-1.14`，再次`catkin_make`工作空间，之后就能正常运行了。
由于记不清我是不是按照郑博在README文件中的顺序安装相关库的，在此强烈建议还没有测试FAST-LIVO* 的朋友们按照顺序安装相关库

也许这些方法仅对我个人有效，但还是希望能帮到大家：）

不知道这种问题出现的原因是什么，希望有朋友可以指点指点：）

PS:虽说代码到最后是冷冰冰的二进制吧，但是第二次经历让我感觉它也挺会“失去了才知道珍惜”：）

maaayz87, 2025-03-25:
同样问题 换了一个opencv版本：4.6.0改成3.4.3. 其他：ubuntu20.04 pcl-1.13 sophus-3.26.6 eigen-3.4.0 可以正常运行

zhao-zhibo, 2025-04-29:
@prexhu @maaayz87  你好，我想问我现在是LIO-SAM，然后也报这个错误( https://github.com/JokerJohn/LIO_SAM_6AXIS/issues/106 )，我该如何进行操作呢？也像你这样catkin_make，然后重新安装pcl吗，我现在试了好几个版本的gtsam 和Eigen，都没有解决这个问题，我之前这个代码还能正常运行，这几天环境可能是变化了，导致代码无法运行了。请问有没有什么建议呢？

maaayz87, 2025-04-29:
您好，我最近正在xxx中，无法亲自回复您的邮件。我将在xx结束后，尽快给您回复。